bitsy-ai/printnanny-os #183: File permission issues on new install

derekslenk, 2022-12-30:
## **Describe the bug**
On a fresh install of v0.5.0, various functions that relate to saving settings (changing camera, editing the settings file) do not work.

Upon further inspection in the console, it looks like there are file permission issues
```
{subject_pattern: 'pi.{pi_id}.settings.camera.apply', request: {…}, error: "failed to make directory '/home/printnanny/.config/printnanny/vcs': Permission denied; class=Os (2)"}
error: "failed to make directory '/home/printnanny/.config/printnanny/vcs': Permission denied; class=Os (2)"
``` 
## **To Reproduce**
On a brand new install of `0.5.0` -
1. Go to 'Edit Settings File'
2. Notice that nothing loads
3. Open the console
4. See error below

## **Expected behavior**
That I would be able to at least see the settings file, ideally editing as the title suggests

## **Screenshots**
[image]

## Issue.txt
Copy and paste the contents of `issue.txt` below.
```
/etc/issue

-----------------------
Build Info:  |
-----------------------
IMAGE_NAME = printnanny-release-image-raspberrypi4-64-20221227205349
DISTRO_PRETTY = PrintNanny Linux 0.5.0 (Emerald)
VARIANT_NAME = PrintNanny OS Core
VARIANT_ID = core
DISTRO = printnanny
DISTRO_NAME = PrintNanny Linux
DISTRO_VERSION = 0.5.0
DISTRO_CODENAME = Emerald
ID_LIKE = BitsyLinux
HOME_URL = https://printnanny.ai
BUG_REPORT_URL = https://github.com/bitsy-ai/printnanny-os/issues
YOCTO_VERSION = 4.0.2
YOCTO_CODENAME = Langdale
MACHINE = raspberrypi4-64
TUNE_PKGARCH = cortexa72
-----------------------
Layer Revisions:      |
-----------------------
meta              = langdale:a3e3b740e140d036122f7b11e2ac452bda548444
meta-poky         = langdale:a3e3b740e140d036122f7b11e2ac452bda548444
meta-yocto-bsp    = langdale:a3e3b740e140d036122f7b11e2ac452bda548444
meta-raspberrypi  = libcamera-v0.0.3:104a8b9290ab4da92c8f693e2fdc42a6396975d2
meta-oe           = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-python       = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-multimedia   = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-bitsy        = main:86c39e7484c178b92c660f2411b07ff46a5fd9e6
meta-neural-network = master:fed68f0a659c9b6431985c156141dc28dc4c2275
meta-printnanny   = main:86c39e7484c178b92c660f2411b07ff46a5fd9e6
meta-networking   = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-filesystems  = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-initramfs    = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-webserver    = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-swupdate     = master:5e611f44db1b9333342ed61d77a3c3ed7be69c2b
workspace         = langdale:a3e3b740e140d036122f7b11e2ac452bda548444
meta-microcontroller = master:9f396e595b2d639be7f2b494f4a61e48d9495058
/etc/os-release

ID=printnanny
ID_LIKE="BitsyLinux"
BUILD_ID="2022-12-21T12:36:49Z"
NAME="PrintNanny Linux"
VERSION="0.5.0 (Emerald)"
VERSION_ID=0.5.0
PRETTY_NAME="PrintNanny Linux 0.5.0 (Emerald)"
VERSION_CODENAME="Emerald"
HOME_URL="https://printnanny.ai"
BUG_REPORT_URL="https://github.com/bitsy-ai/printnanny-os/issues"
PRIVACY_POLICY_URL="https://printnanny.ai/privacy-policy"
SUPPORT_URL="https//printnanny.ai"
YOCTO_VERSION="4.0.2"
YOCTO_CODENAME="Langdale"
SDK_VERSION="0.5.0"
```

NB: If I `chown -R printnanny:printnanny /home/printnanny/.config/printnanny` that I can edit the settings file

leigh-johnson, 2022-12-31:
:tada: Fixed this in the latest nightly. I'll close out this issue with a link to the v0.5.1 release when it's available. 

```
pn-v0-5:~$ ls -ahl /home/printnanny/
total 17K
drwxrwxr-x 1 printnanny printnanny 1.0K Dec 31 14:26 .
drwxr-xr-x 1 root       root       1.0K Nov  8 06:00 ..
-rwxrwxr-x 1 printnanny printnanny  410 Mar  9  2018 .bashrc
drwxrwxr-x 1 printnanny printnanny 1.0K Dec 31 14:26 .config
drwxrwxr-x 1 printnanny printnanny 1.0K Mar  9  2018 .klipper
drwxrwxr-x 1 printnanny printnanny 1.0K Mar  9  2018 .local
drwxrwxr-x 1 printnanny printnanny 1.0K Mar  9  2018 .moonraker
-rwxrwxr-x 1 printnanny printnanny  241 Mar  9  2018 .profile
drwxr-xr-x 3 printnanny printnanny 1.0K Dec 31 14:26 Sync
drwxr-xr-x 5 printnanny printnanny 1.0K Dec 31 14:26 klipper-venv
drwxr-xr-x 5 printnanny printnanny 1.0K Dec 31 14:26 moonraker-venv
drwxr-xr-x 5 printnanny printnanny 1.0K Dec 31 14:26 octoprint-venv
```

derekslenk, 2023-01-02:
I have really wanted things to work so I did an install of 0.4.5 and then used the swu file to go to 0.5.0.

First error I had is that `/run` wasn't big enough for the update file (where nginx stores it)

And then after the upgrade, when trying to go to the settings file, I still get: `error: "failed to make directory '/home/printnanny/.config/printnanny/vcs': Permission denied; class=Os (2)"`

So it might be affecting other installs as well?

leigh-johnson, 2023-01-07:
👀 Closing this out, since it looks like this is no longer happening to you in v0.5.1 - but please feel free to re-open if you are still impacted by file permissions issues. Going through your bug list now, thank you!

closed 2023-01-07 by leigh-johnson