bitsy-ai/printnanny-os #193: FilamentManager plugin install fails to start

leigh-johnson, 2023-01-07:
## **Describe the bug**
The filament manager plugin fails to load with error:
```
Jan 07 14:07:03 pn-v0-5 octoprint[1041]: 2023-01-07 14:07:03,738 - octoprint.plugin.core - ERROR - Error loading plugin filamentmanager
Jan 07 14:07:03 pn-v0-5 octoprint[1041]: Traceback (most recent call last):
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "/usr/lib/python3.10/site-packages/octoprint/plugin/core.py", line 1274, in _import_plugin
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:     module = _load_module(module_name, spec)
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "/usr/lib/python3.10/site-packages/octoprint/plugin/core.py", line 52, in _load_module
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:     return imp.load_module(name, f, filename, details)
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "/usr/lib/python3.10/site-packages/octoprint/vendor/imp.py", line 238, in load_module
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:     return load_package(name, filename)
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "/usr/lib/python3.10/site-packages/octoprint/vendor/imp.py", line 212, in load_package
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:     return _load(spec)
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "<frozen importlib._bootstrap>", line 719, in _load
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "/home/printnanny/octoprint-venv/lib/python3.10/site-packages/octoprint_filamentmanager/__init__.py", line 17, in <module>
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:     from .data import FilamentManager
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "/home/printnanny/octoprint-venv/lib/python3.10/site-packages/octoprint_filamentmanager/data/__init__.py", line 13, in <m>
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:     from sqlalchemy.engine.url import URL
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "/home/printnanny/octoprint-venv/lib/python3.10/site-packages/sqlalchemy/__init__.py", line 9, in <module>
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:     from .sql import (
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "/home/printnanny/octoprint-venv/lib/python3.10/site-packages/sqlalchemy/sql/__init__.py", line 8, in <module>
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:     from .expression import (
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "/home/printnanny/octoprint-venv/lib/python3.10/site-packages/sqlalchemy/sql/expression.py", line 34, in <module>
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:     from .functions import func, modifier, FunctionElement, Function
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "/home/printnanny/octoprint-venv/lib/python3.10/site-packages/sqlalchemy/sql/functions.py", line 11, in <module>
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:     from . import sqltypes, schema
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "/home/printnanny/octoprint-venv/lib/python3.10/site-packages/sqlalchemy/sql/sqltypes.py", line 17, in <module>
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:     from . import elements
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "/home/printnanny/octoprint-venv/lib/python3.10/site-packages/sqlalchemy/sql/elements.py", line 16, in <module>
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:     from . import type_api
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "/home/printnanny/octoprint-venv/lib/python3.10/site-packages/sqlalchemy/sql/type_api.py", line 16, in <module>
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:     from .base import SchemaEventTarget
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:   File "/home/printnanny/octoprint-venv/lib/python3.10/site-packages/sqlalchemy/sql/base.py", line 49, in <module>
Jan 07 14:07:03 pn-v0-5 octoprint[1041]:     class _DialectArgView(collections.MutableMapping):
Jan 07 14:07:03 pn-v0-5 octoprint[1041]: AttributeError: module 'collections' has no attribute 'MutableMapping'
```

https://plugins.octoprint.org/plugins/filamentmanager/

## **To Reproduce**
What are the steps to reproduce the issue?
1. Install FilamentManager plugin (OctoPrint settings menu)
2. Notice: plugin throws uncaught exception 

## Issue.txt

```
-----------------------
Build Info:  |
-----------------------
IMAGE_NAME = printnanny-release-image-raspberrypi4-64-20230106203240
DISTRO_PRETTY = PrintNanny Linux 0.5.1 (Emerald)
VARIANT_NAME = PrintNanny OS Core
VARIANT_ID = core
DISTRO = printnanny
DISTRO_NAME = PrintNanny Linux
DISTRO_VERSION = 0.5.1
DISTRO_CODENAME = Emerald
ID_LIKE = BitsyLinux
HOME_URL = https://printnanny.ai
BUG_REPORT_URL = https://github.com/bitsy-ai/printnanny-os/issues
YOCTO_VERSION = 4.0.2
YOCTO_CODENAME = Langdale
MACHINE = raspberrypi4-64
TUNE_PKGARCH = cortexa72
-----------------------
Layer Revisions:      |
-----------------------
meta              = langdale:a3e3b740e140d036122f7b11e2ac452bda548444
meta-poky         = langdale:a3e3b740e140d036122f7b11e2ac452bda548444
meta-yocto-bsp    = langdale:a3e3b740e140d036122f7b11e2ac452bda548444
meta-raspberrypi  = libcamera-v0.0.3:104a8b9290ab4da92c8f693e2fdc42a6396975d2
meta-oe           = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-python       = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-multimedia   = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-bitsy        = main:957dfaa51204ce7739a3d222e50c5c33d656481e
meta-neural-network = master:fed68f0a659c9b6431985c156141dc28dc4c2275
meta-printnanny   = main:957dfaa51204ce7739a3d222e50c5c33d656481e
meta-networking   = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-filesystems  = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-initramfs    = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-webserver    = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-swupdate     = master:5e611f44db1b9333342ed61d77a3c3ed7be69c2b
workspace         = langdale:a3e3b740e140d036122f7b11e2ac452bda548444
meta-microcontroller = master:9f396e595b2d639be7f2b494f4a61e48d9495058
```

[install.log](https://github.com/bitsy-ai/printnanny-os/files/10367276/install.log)

leigh-johnson, 2023-01-07:
It looks like `collections.MutableMapping` was moved to `collections.abc.MutableMapping` in Python versions 3.10 and above. I should be able to submit a patch to the FilamentManager plugin, and I'll provide steps to install from my branch as a workaround. 

https://stackoverflow.com/a/71902541

leigh-johnson, 2023-01-07:
Just as a note, we'll run into this issue :point_down: if we ever upgrade pip to >=23.1 
https://github.com/OllisGit/OctoPrint-FilamentManager/issues/108

We're on pip 22.2.x in the Yocto Langdale releases of PrintNanny OS, but I figured I'd jot this down as a gift to future-me. 
```pip3 --version
pip 22.2.2 from /usr/lib/python3.10/site-packages/pip (python 3.10)
```

leigh-johnson, 2023-01-08:
:tada: I got the install working! 

https://github.com/bitsy-ai/OctoPrint-FilamentManager/zipball/fix-python-10-compat

Here's the way to install this if you're using v0.5.1:

1. Open OctoPrint Settings menu
2. Open the "Plugin Manager" section
3. Click the "+Get More" button in the upper-right 
4. Enter the following URL in the "... from URL" input box, then click the "Install" button
https://github.com/bitsy-ai/OctoPrint-FilamentManager/zipball/fix-python-10-compat
5. Restart OctoPrint after the install finished (you'll be prompted to do this)

If the plugin seems to work as expected, I can bake this patch into PrintNanny OS while we wait for the maintainer to check out my PR. :dancing_women:

derekslenk, 2023-01-08:
Works like a champ!

closed 2023-01-09 by leigh-johnson