bitsy-ai/printnanny-os #219: PrintNanny camera settings shows IMX219 camera in drop-down menu when only USB camera is connected

leigh-johnson, 2023-01-25:
## **Describe the bug**
When only a USB camera is connected, the IMX219 camera sensor (CSI ribbon cable) still appears in the drop-down menu of PrintNanny's camera settings menu.

When only a USB camera is present, the USB camera should be the default.

## Issue.txt

```
-----------------------
Build Info:  |
-----------------------
IMAGE_NAME = printnanny-release-image-raspberrypi4-64-20230124200428
DISTRO_PRETTY = PrintNanny Linux 0.6.0 (Feldspar)
VARIANT_NAME = PrintNanny OS Core
VARIANT_ID = core
DISTRO = printnanny
DISTRO_NAME = PrintNanny Linux
DISTRO_VERSION = 0.6.0
DISTRO_CODENAME = Feldspar
ID_LIKE = BitsyLinux
HOME_URL = https://printnanny.ai
BUG_REPORT_URL = https://github.com/bitsy-ai/printnanny-os/issues
YOCTO_VERSION = 4.0.2
YOCTO_CODENAME = Langdale
MACHINE = raspberrypi4-64
TUNE_PKGARCH = cortexa72
-----------------------
Layer Revisions:      |
-----------------------
meta              = langdale:a3e3b740e140d036122f7b11e2ac452bda548444
meta-poky         = langdale:a3e3b740e140d036122f7b11e2ac452bda548444
meta-yocto-bsp    = langdale:a3e3b740e140d036122f7b11e2ac452bda548444
meta-raspberrypi  = libcamera-v0.0.3:104a8b9290ab4da92c8f693e2fdc42a6396975d2
meta-oe           = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-python       = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-multimedia   = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-bitsy        = printnanny-snapshot-service:ae6710e5858ac0b13cd10c89ead66ba80537c6ae
meta-neural-network = master:fed68f0a659c9b6431985c156141dc28dc4c2275
meta-printnanny   = printnanny-snapshot-service:ae6710e5858ac0b13cd10c89ead66ba80537c6ae
meta-networking   = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-filesystems  = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-initramfs    = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-webserver    = langdale:c5668905a6d8a78fb72c2cbf8b20e91e686ceb86
meta-swupdate     = master:5e611f44db1b9333342ed61d77a3c3ed7be69c2b
workspace         = langdale:a3e3b740e140d036122f7b11e2ac452bda548444
meta-microcontroller = master:9f396e595b2d639be7f2b494f4a61e48d9495058
```

leigh-johnson, 2023-01-28:
I'm pushing this into v0.6.2 - I need to figure out a better way to scan for the default value here.

I'm still seeing the default IMX219 camera, even when only my USB brio is plugged in:

[image]

leigh-johnson, 2023-01-28:
So, I'm seeing two separate issues here:

1. `"Raspberry Pi imx219"` label is being applied to a device with name `/base/scb/pcie@7d500000/pci@0,0/usb@0,0-1.3:1.0-046d:085e`

Our hotplug check should also update the label field:
https://github.com/bitsy-ai/printnanny-rs/blob/cc8e3e5df8f88bf2718ab8488898682853248dff/settings/src/cam.rs#L474

2. Caps aren't being negotiated correctly for the USB camera:

```
Jan 27 18:56:17 printnanny gstd[2868]: [0:31:56.092870617] [2904]  INFO Camera camera.cpp:1026 configuring streams: (0) 640x480-YUYV
Jan 27 18:56:17 printnanny gstd[2868]: 0:00:23.790196042  2868   0x7f9804fc60 WARN                 default gstlibcamera-utils.cpp:441:gst_libcamera_clamp_and_set_frameduration: FrameDurationLimits not found in camera controls.
Jan 27 18:56:17 printnanny gstd[2868]: 0:00:23.791020449  2868   0x55a44c0400 WARN                    v4l2 gstv4l2object.c:4571:gst_v4l2_object_probe_caps:<v4l2h264enc0:src> Failed to probe pixel aspect ratio with VIDIOC_CROPCAP: Invalid argument
Jan 27 18:56:17 printnanny gstd[2868]: 0:00:23.791554893  2868   0x7f9804fc60 WARN                GST_CAPS gstpad.c:5757:pre_eventfunc_check:<capsfilter0:sink> caps video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, colorimetry=(string)2:4:5:1, framerate=(fraction)16/1 not accepted
Jan 27 18:56:17 printnanny gstd[2868]: 0:00:23.792154264  2868   0x7f9804fc60 WARN                GST_CAPS gstpad.c:5757:pre_eventfunc_check:<capsfilter0:sink> caps video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, colorimetry=(string)2:4:5:1, framerate=(fraction)16/1 not accepted
Jan 27 18:56:17 printnanny gstd[2868]: 0:00:23.803963986  2868   0x7f98065e40 FIXME                default gstutils.c:4025:gst_pad_create_stream_id_internal:<h264_src:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
Jan 27 18:56:17 printnanny gstd[2868]: 0:00:23.808415579  2868   0x55a44c0760 FIXME                default gstutils.c:4025:gst_pad_create_stream_id_internal:<rtp_src:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
Jan 27 18:56:17 printnanny gstd[2868]: 0:00:23.812435375  2868   0x7f98065d20 FIXME                default gstutils.c:4025:gst_pad_create_stream_id_internal:<tflite_inference_src:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
Jan 27 18:56:17 printnanny gstd[2868]: 0:00:23.821498838  2868   0x55a44c0400 WARN                    v4l2 gstv4l2object.c:4571:gst_v4l2_object_probe_caps:<v4l2h264enc1:src> Failed to probe pixel aspect ratio with VIDIOC_CROPCAP: Invalid argument
Jan 27 18:56:17 printnanny gstd[2868]: 0:00:23.828863338  2868   0x7f9804f520 FIXME                default gstutils.c:4025:gst_pad_create_stream_id_internal:<bounding_boxes_src:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
Jan 27 18:56:18 printnanny gstd[2868]: 0:00:24.334454208  2868   0x7f98072d20 FIXME                default gstutils.c:4025:gst_pad_create_stream_id_internal:<df_src:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
Jan 27 18:56:18 printnanny gstd[2868]: 0:00:24.590538374  2868   0x7f9826eb60 FIXME                default gstutils.c:4025:gst_pad_create_stream_id_internal:<hls_src:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
Jan 27 18:56:18 printnanny gstd[2868]: 0:00:24.837833189  2868   0x55a44c0400 WARN                    v4l2 gstv4l2object.c:4571:gst_v4l2_object_probe_caps:<v4l2jpegenc0:src> Failed to probe pixel aspect ratio with VIDIOC_CROPCAP: Invalid argument
Jan 27 18:56:18 printnanny gstd[2868]: 0:00:24.839342411  2868   0x7f9826f000 FIXME                default gstutils.c:4025:gst_pad_create_stream_id_internal:<snapshot_src:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
Jan 27 18:56:18 printnanny gstd[2868]: [0:31:57.219657801] [2903]  INFO V4L2 v4l2_videodevice.cpp:1820 /dev/video0[35:cap]: Zero sequence expected for first frame (got 2)
Jan 27 18:56:18 printnanny gstd[2868]: 0:00:24.916403263  2868   0x7f9804fc60 WARN                GST_CAPS gstpad.c:5757:pre_eventfunc_check:<capsfilter0:sink> caps video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, colorimetry=(string)2:4:5:1, framerate=(fraction)16/1 not accepted
Jan 27 18:56:18 printnanny gstd[2868]: 0:00:24.916586856  2868   0x7f9804fc60 WARN            libcamerasrc gstlibcamerasrc.cpp:312:processRequest:<libcamerasrc0> error: Internal data stream error.
Jan 27 18:56:18 printnanny gstd[2868]: 0:00:24.916637022  2868   0x7f9804fc60 WARN            libcamerasrc gstlibcamerasrc.cpp:312:processRequest:<libcamerasrc0> error: streaming stopped, reason not-negotiated (-4)
Jan 27 18:56:19 printnanny gstd[2868]: 0:00:24.938952485  2868   0x7f9804f520 WARN           basetransform gstbasetransform.c:1371:gst_base_transform_setcaps:<tensordecoder0> transform could not transform other/tensors, format=(string)static, num_tensors=(int)4, framerate=(fraction)0/1, dimensions=(string)"4:40:1:1\,40:1:1:1\,40:1:1:1\,1:1:1:1", types=(string)"float32\,float32\,float32\,float32" in anything we support
```

The caps available on my Logitech Brio are:

```
rintnanny:~$ gst-device-monitor-1.0  
Probing devices...

[... cut ...]
	       video/x-raw, format=YUY2, width=848, height=480
	       video/x-raw, format=YUY2, width=800, height=600
	       video/x-raw, format=YUY2, width=960, height=540
	       video/x-raw, format=YUY2, width=1024, height=576
	       video/x-raw, format=YUY2, width=1280, height=720
	       video/x-raw, format=YUY2, width=1600, height=896
	       video/x-raw, format=YUY2, width=1920, height=1080
	gst-launch-1.0 libcamerasrc camera-name="/base/scb/pcie\@7d500000/pci\@0\,0/usb\@0\,0-1.3:1.0-046d:085e" ! ...
```

leigh-johnson, 2023-01-28:
I'm pulling https://github.com/bitsy-ai/meta-bitsy/pull/393 into v0.6.1 - this just increases the verbosity of Gstreamer caps negotiation and src/sink pad probing, so I'll be able to see more info in a crash report.

derekslenk, 2023-01-29:
I have more cameras I can test with as well. That camera is a Logitech c290. I have a Logitech c920 I could test, and I also have a SJCAM that has a webcam mode that works with at least Win11 with no additional drivers

leigh-johnson, 2023-01-29:
> I have more cameras I can test with as well. That camera is a Logitech c290. I have a Logitech c920 I could test, and I also have a SJCAM that has a webcam mode that works with at least Win11 with no additional drivers

That'd be great! I increased logging verbosity, so all you have to do is submit a crash report if your c920 or SJCAM aren't working on v0.6.2. 

Compatibility comes down to Gstreamer element capability negotiation ("caps negotiation"), rather than drivers. Here's an example of a failure I'm seeing on my Brio at the moment: 

```
Jan 28 17:52:08 printnanny gstd[384]: 0:00:38.427564514   384   0x7f8824c400 WARN                GST_CAPS gstpad.c:3235:gst_pad_query_accept_caps_default:<v4l2jpegenc0:sink> caps: video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)16/1, format=(string)YUY2, colorimetry=(string)bt709 were not compatible with: video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)16/1, format=(string)YUY2, colorimetry=(string)bt709, interlace-mode=(string)progressive
Jan 28 17:52:08 printnanny gstd[384]: 0:00:38.427706940   384   0x7f8824c400 WARN                GST_CAPS gstpad.c:5757:pre_eventfunc_check:<v4l2jpegenc0:sink> caps video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)16/1, format=(string)YUY2, colorimetry=(string)bt709 not accepted
```
:eyes: Looking into why the upstream buffer isn't negotiating `interlace-mode=(string)progressive` now

leigh-johnson, 2023-01-31:
@derekslenk When you get a chance, can you run the following command on a Raspberry Pi with your USB camera connected?

```
$ gst-device-monitor-1.0 Source/Video
```

This will output the capabilities of your camera (resolution, pixel format, colorimetry, etc). Thank you!

leigh-johnson, 2023-02-01:
:+1: Verified this is fixed in the v0.6.2 nightly

closed 2023-02-01 by leigh-johnson