bitsy-ai/printnanny-os #264: SSH Terminal window

georgera, 2023-02-11:
Enhancement to consider:

For users who are not familiar with Linux or have a cmd line fear; is there an open-source SSH terminal window you can include in PNOS main menu that will open a terminal window and SSH session with the host?

Just a thought.   It may help you support those who are not comfortable with the Penguin.

leigh-johnson, 2023-02-15:
👋 What are some tasks you're trying to accomplish?

I've considered Apache Guacamole for remote SSH/RDP. The technology stack is similar if you've ever used an in-browser shell prompt from AWS, GCP, Azure, etc. The downside is that the Guacamole stack adds many new attack surfaces to PNOS. 

https://guacamole.apache.org/

I have yet to pull the trigger on this because I want to encourage people to use PrintNanny's workflow system to accomplish most remote control/automation tasks. My gut instinct is script snippets provided with https://www.windmill.dev/ will be more useful than a remote shell prompt for users without Linux experience.

georgera, 2023-02-16:
Understand, don’t want to open any security holes.   It was just a thought .   

 

I don’t have immediate need, just thought it might make it easier for non-linux users.   I don’t mind using Terminal on mac. 

 

Fraternally, 

 

 

Ill. Sir Russell Georgen
Past Potentate 2020
Moolah Temple

 

 

👋 What are some tasks you're trying to accomplish?

I've considered Apache Guacamole for remote SSH/RDP. The technology stack is similar if you've ever used an in-browser shell prompt from AWS, GCP, Azure, etc. The downside is that the Guacamole stack adds many new attack surfaces to PNOS.

https://guacamole.apache.org/

I have yet to pull the trigger on this because I want to encourage people to use PrintNanny's workflow system to accomplish most remote control/automation tasks. My gut instinct is script snippets provided with https://www.windmill.dev/ will be more useful than a remote shell prompt for users without Linux experience.