bitsy-ai/printnanny-os #278: Every thing just stopped?

georgera, 2023-02-23:
## **Describe the bug**
I looked over at printer and everything just stopped?   Can't give you a better explanation.   

## **To Reproduce**
What are the steps to reproduce the issue?
Plan not sure,  I was playing around with Camera view and sent you a dump of it not being able to restart.   It may have happened after

## **Expected behavior**
mmmmmm,   it shouldn't have stopped?   ;-)   Just kidding. 

## **Screenshots**
I will attach or email you video. 

## Issue.txt
Copy and paste the contents of `issue.txt` below.
```
# paste issue.txt here
```

> ℹ️ You can find `/boot/issue.txt` on your SD card. You can also find the contents of `issue.txt` in OctoPrint's Settings > OctoPrint-Nanny > Report Issue tab.

leigh-johnson, 2023-02-23:
Hi @georgera, I saw that submitting a crash report via Mission Control failed for you. Can you please submit a crash report using the command-line tool instead? I'll look into what happened here.

To submit a crash report via command-line, ssh into your Raspberry Pi and run the following:
```
$ sudo -u printnanny printnanny crash-report
```

leigh-johnson, 2023-02-25:
Closing this issue - I think it's related to https://github.com/bitsy-ai/printnanny-os/issues/282

If Wifi disconnects, OctoPrint will restart. This would interrupt a print job. I'll track patches for this in #282

closed 2023-02-25 by leigh-johnson